intelligent-agent/redeem #1: Unwanted change in homing behavior

TechMav75, 2017-09-16:
Versions:
Octoprint 1.3.4
Redeem 0129bd0204f70648ff8cb48cb5f6e1e778ba655b
Toggle 607ac355d625ea3077b472dc4cc08dac45d68bd2

Slice my file, upload the GCode, hit print
It immediately zeros the Z axis and leaves the bed at 0
Then heats the bed, then the hot end, then homes on X and Y, dragging the tip across the bed
THEN it rezero's the Z axis, lifts 5mm (like it should) then starts the print

Previously, it would sit wherever it was and preheat the bed and hot end, then lower the bed, home X/Y, then home Z, lift 5mm, and start the print

Only GCode script in place for 'Before Print Starts' is G28 Z5 F240

eliasbakken, 2017-09-16:
It's a bit difficult to understand what is expected behavior and not in this scenario. Could you clarify? Also, could you post the first lines of the G-code or your "start-gcode" from your slicer?

TechMav75, 2017-09-16:
Expected behavior is as above, under 'Previously'
When I would hit print, it would preheat the bed and nozzle, then drop the bed 5mm, home X/Y, then home Z.

GCode file head (first 30 lines)
M190 S50.000000
M109 S230.000000
;Basic settings: Layer height: 0.1 Walls: 1 Fill: 50
;Print time: 2 hours 25 minutes
;Filament used: 12.843m 38.0g
;Filament cost: None
;M190 S50 ;Uncomment to add your own bed temperature line
;M109 S230 ;Uncomment to add your own temperature line
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G1 Z15.0 F6000 ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E10             ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F6000
;Put printing message on LCD screen
M117 Printing...
;Layer count: 40
;LAYER:0
M107
G0 F6000 X86.649 Y66.517 Z0.100
;TYPE:SKIRT
G1 F3000 X113.220 Y27.624 E0.93020

Only GCode script in place for 'Before Print Starts' is  G28 Z5 F240
Which drops the bed 5mm before it moves to start the print.

TechMav75, 2017-09-17:
Tracked this down to an issue in the slicer. I apologize for wasting your time.

closed 2017-09-17 by TechMav75