intelligent-agent/redeem #13: G28 as a macro

eliasbakken, 2017-09-27:
For printers using only the Z probe for Z endstop, it's important to be able to inject G-code before homing occurs. On Marlin, a safe Z is defined while in RepRap firmware, macros for different printers are defined.

Wackerbarth, 2018-02-24:
I like the idea of having the option for the user to define his own sequence (a macro). However, to do this, we need to be able to have a "return to axis end stop" as a code different from the general G28 which would invoke it.  In my case, I have a mechanical Z end stop switch which is below the bed. Thus, I want to perform the following sequence:

1. Raise the extruder so that it doesn't dig into the bed
2. Move the head to the X&Y mechanical stops.
3. Reposition the head so that it will clear some obstructions
4. Move the head to the Z mechanical stop
5. Reposition the head to a "parked" position

I currently do this as a part of my "G29" sequence using "G28" in steps 2 and 4.
If we want to be able to implement "G28" in a similar manner,  it would be problematic because we do not have a code for those motions.
Any suggestions on how we might express the action?

manbehindthemadness, 2019-10-15:
My though is rather than adding a macro to G28, create a handler that would allow a macro to be attached to any gcode statement, It strikes me that would be about the same amount of work whilst offering much more long-term flexibility.

Wackerbarth, 2019-10-16:
@manbehindthemadness -- I agree that this would be a desirable approach. But, as I point out above, we really need to decouple the primitive machine operations from the command instructions (g-code) that activate them. Then we define an intermediate language (the macro definition language) that "compiles" a sequence (often only one "instruction") of those primitive operations given a command instruction. This can be accomplished both simply and efficiently by moving the implementation procedure names to a different namespace and setting up (at initialization) the trivial mapping from the external namespace to the intermediate one. The basic mechanism is already present in our command processing. It just changes the perspective that a "macro" is the generic case rather than being a special case.