intelligent-agent/redeem #21: G1: 'Stepper_004A' object has no attribute 'current_enabled'

carlsoar, 2017-10-11:
*Copied from Bitbucket issues list*

 Boris Lašič created an issue 2017-09-04
It seems that Older Stepper classes were not updated when introducing revision B.
Stepper_00A4 does not implement set_current_disabled/ set_current_enabled function and does not have a current_enabled property.
That makes redeem unusable on rev A4 replicapes.
Applies to kamikaze image 2.1 and develop branch.
Comments (1)
 Daryl Bond
I can confirm this.
an hour ago

zaped212, 2019-01-26:
Should this issue be closed?
I am running the latest version of the RC, and have a A4A board and have confirmed that the support has been added to the RC branch.

goeland86, 2019-01-27:
I think you're right @zaped212 - we can see that @darylbond actually fixed the problem over a year ago, too.

closed 2019-01-27 by goeland86