intelligent-agent/redeem #34: Behavior with printing from SD card.

RaveGun, 2017-11-04:


RaveGun, 2017-11-04:
Output from Reedem after starting and stopping a print from SD card:
```
11-04 12:03 root         INFO     M23: starting gcode file processing
11-04 12:03 root         ERROR    Error while executing M23: coercing to Unicode: need string or buffer, NoneType found
11-04 12:03 root         ERROR    Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Redeem-2.0.5-py2.7-linux-armv7l.egg/redeem/GCodeProcessor.py", line 126, in execute
    self.gcodes[val].execute(gcode)
  File "/usr/local/lib/python2.7/dist-packages/Redeem-2.0.5-py2.7-linux-armv7l.egg/redeem/gcodes/M2x.py", line 267, in execute
    for root, directories, filenames in os.walk(list_location):
  File "/usr/lib/python2.7/os.py", line 278, in walk
    names = listdir(top)
TypeError: coercing to Unicode: need string or buffer, NoneType found

11-04 12:03 root         INFO     M24: current file is: '/usr/share/models/Soap_Holder_v4.gcode'
11-04 12:03 root         INFO     M24: active file is '/usr/share/models/Soap_Holder_v4.gcode'
11-04 12:03 root         INFO     M24: file open: '/usr/share/models/Soap_Holder_v4.gcode'
11-04 12:04 root         ERROR    No GCode processor for M26. Message: M26 S0

```

RaveGun, 2017-11-04:
After a disconnect - connect from Octoprint and trying to move the nozzle up redeem stopped and throwed:
`        [ 1509797528860 ]       queueMove FAILED: axis 0 end position outside of soft limit (end = 0.2, max limit = 0.085)
terminate called after throwing an instance of 'Swig::DirectorMethodException'
  what():  SWIG director method error. Error detected when calling 'AlarmCallbackNative.call'
Aborted`

darylbond, 2017-11-04:
The out of soft limit crash is unrelated to the SD card implementation and should probably be listed under a separate issue.

ajmirsky, 2017-11-06:
@RaveGun @darylbond soft limit crash is same as #14

goeland86, 2017-11-20:
@RaveGun do you still have the issue?

Regarding the long-running gcode, add the M23 and M24 commands to the list in OctoPrint settings (settings -> serial connection -> advanced settings (at the bottom) )

Try pulling the latest version of develop to test it out.

RaveGun, 2017-11-20:
I cannot install the updated Redeem.
It is throwing something like:
`Searching for numpy==1.13.0
Reading https://pypi.python.org/simple/numpy/
Best match: numpy 1.13.0
Downloading https://pypi.python.org/packages/05/84/0feb999c05f252af50a5fbc463268044feda92cdaad8cb0d0a6073d76057/numpy-1.13.0.zip#md5=fd044f0b8079abeaf5e6d2e93b2c1d03
Processing numpy-1.13.0.zip
Writing /tmp/easy_install-cocu2n/numpy-1.13.0/setup.cfg
Running numpy-1.13.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-cocu2n/numpy-1.13.0/egg-dist-tmp-2iIkG7
/tmp/easy_install-cocu2n/numpy-1.13.0/setup.py:367: UserWarning: Unrecognized setuptools command, proceeding with generating Cython sources and expanding templates
  run_build = parse_setuppy_commands()
/usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
  warnings.warn(msg)
Traceback (most recent call last):
  File "setup.py", line 107, in <module>
    'redeem = redeem.Redeem:main'
  File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
....
....
  File "/usr/lib/python2.7/dist-packages/numpy/distutils/misc_util.py", line 1003, in add_subpackage
    caller_level = 2)
  File "/usr/lib/python2.7/dist-packages/numpy/distutils/misc_util.py", line 972, in get_subpackage
    caller_level = caller_level + 1)
  File "/usr/lib/python2.7/dist-packages/numpy/distutils/misc_util.py", line 884, in _get_configuration_from_setup_py
    ('.py', 'U', 1))
  File "numpy/core/setup.py", line 13, in <module>

ImportError: No module named _build_utils.apple_accelerate`

Pip list shows I got numpy == 1.11.0.

goeland86, 2017-11-20:
Ah, that's... unfortunate, but easily solved. `pip install --upgrade numpy` should do the trick. Check that you don't also need sh and docutils?

RaveGun, 2017-11-20:
I changed the requirements.txt to numpy 1.11.0.
After this it was whining about Adafruit_GPIO.
Now I am trying the `pip install --upgrade numpy` that it takes ages.
`root@kamikaze:/usr/src/redeem# pip install --upgrade numpy
Collecting numpy
  Using cached numpy-1.13.3.zip
Building wheels for collected packages: numpy
  Running setup.py bdist_wheel for numpy ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Ctp7GU/numpy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /tmp/tmpudRtI5pip-wheel- --python-tag cp27:
  Running from numpy source directory.
  /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
    warnings.warn(msg)
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for numpy
  Running setup.py clean for numpy
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Ctp7GU/numpy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" clean --all:
  Running from numpy source directory.

  `setup.py clean` is not supported, use one of the following instead:

    - `git clean -xdf` (cleans all files)
    - `git clean -Xdf` (cleans all versioned files, doesn't touch
                        files that aren't checked into the git repo)

  Add `--force` to your command to use it anyway if you must (unsupported).

  ----------------------------------------
  Failed cleaning build dir for numpy
Failed to build numpy
Installing collected packages: numpy
  Found existing installation: numpy 1.11.0
    Not uninstalling numpy at /usr/lib/python2.7/dist-packages, outside environment /usr
  Running setup.py install for numpy ... \`

I tried already to upgrade the `pip` but this is also not working:
`Not uninstalling pip at /usr/lib/python2.7/dist-packages, outside environment /usr`

goeland86, 2017-11-20:
Hmmm. Very odd. I still also run numpy 1.11.0. I'll poke the devs to help me figure this out.

RaveGun, 2017-11-20:
So, after running `easy_install -U pip`, I managed to upgrade pip to 9.0.1.
Now is trying to update numpy to 1.13.3.
I have no idea when it will finish.

RaveGun, 2017-11-21:
Over night it installed numpy 1.13.3. I modified the requirements.txt to match and now it is working and showing the SD files. I did not tested any prints.

ajmirsky, 2017-11-29:
@RaveGun were those updates to `develop` able to fix this issue? 

The DirectorMethodException has been open as a separate issue ( #14 )

RaveGun, 2017-12-02:
Hello,
I am not so sure because I also made some changes on my own. Now it shows the files also on the SD but it does not print. Trying to print does nothing.

I could not follow any further because, after the previous update, the geometry on my delta is now off.
Looks like the old configuration values fit no longer.

RaveGun, 2019-01-16:
Hi,
Wow, this is really old.
No, I am now on Boxing Day and I am able to print from the SD.
Please close the issue.
Thank you.

closed 2019-01-16 by goeland86