intelligent-agent/redeem #46: Ability to control fan based on multiple thermistors

wilhoj, 2017-12-04:
I wanted to connect my both hotends fan to single fan output. (wanted to save one fan output for controlling enclosure temperature) if i connect both with connect-therm-E-fan-2 and connect-therm-H-fan-2 this leads to unstable fan behavior (both processes tries to control fan).

So what i am needing, is that fan is running based on hottest thermistor reading..

I might write this feature, does this rough plan sound reasonable?

- create simple Class which takes as input array of Heaters and spits out hottest temp with 
get_temperature()

-introduce few new config items something like: connect-allhotend-fan-1, connect-allhotend-fan-2 etc..

- if those are enabled on config, on Redeem init creates new object based on this new Class and creates Cooler based on that and appropriate fan..

darylbond, 2018-01-22:
This should now be possible in the develop branch

Wackerbarth, 2018-04-24:
There is an implementation which will enable this functionality (although not exactly as described) which is in testing on the 2.2 development. However, beware that it is still a WIP and may end up in a significantly different form.