intelligent-agent/redeem #51: Options missing from default  configuration

Wackerbarth, 2017-12-30:
While setting up my first Replicape printer, I wanted to activate various features at low temperatures in order to make sure that everything was basically functioning.

I set up a fan to turn on as the hot end  begins to warm.
[Cold-ends]
# If you want the fan-thermitor connetions to have a different temperature:
# therm-e-fan-0-target_temp = 70
therm-e-fan-1-target_temp = 40

It functions just fine. But I see an error in the log ...

12-30 16:34 root         INFO     Using config file /etc/redeem/local.cfg
12-30 16:34 root         WARNING  Option therm-e-fan-1-target_temp in section Cold-ends does not exist in default.cfg
12-30 16:34 root         WARNING  /etc/redeem/local.cfg contains errors.

Now, we could suppress this WARNING by including in default.cfg a bunch of entries of the form 
therm-e-fan-0-target_temp = 60

But that is really a messy way to handle it and it defeats the hard-coded default which is in the code.

So, I would like to see some scheme which will cover the situation in one or two lines

Perhaps, a wildcard description such as
therm-*-fan-*-target_temp = 60

Another approach would be to have the code "verify" that the entry will get referenced somewhere.

Discussion, please.

Wackerbarth, 2018-01-05:
@goeland86 - thanks for the reference to the slack channel. I joined and searched, but found nothing labeled @ajmirsky. I'm guessing that he is @andrewj over there. Will use that channel for discussion and leave this open until we have a "solution" to address the issue.

goeland86, 2018-01-05:
Correct, our usernames are not all the same. For reference I'm @jon_c on Slack.

Wackerbarth, 2018-04-20:
This has been addressed

closed 2018-04-20 by Wackerbarth