intelligent-agent/redeem #59: Buffer Causes Parts Cooling Fan to Run Ahead of Axis Motions Printer

j-pickens, 2018-01-15:
Due to the different way "real time" events are processed from Gcode into actions on the printer, the timing of parts cooling fan events precedes the stepper motor actions of the printer.  This means that the cooling fan is, for instance, cooling at higher rates for bridging operations prior to the actual bridging motions being implemented.  This is exacerbated by having to add buffer delays into the local.cfg file to prevent stuttering motions with Delta printer kinematics when the processor can't keep up with the demanded print moves.

Example buffer settings from my Woodstock Delta Printer:

 [Planner]
max_buffered_move_time =10000
move_cache_size = 4096

goeland86, 2018-01-15:
@j-pickens so there's been a lot of discussion and debate on this - if you take a look at what's happening, (and @ThatWileyGuy did in detail), it was actually OctoPrint not feeding GCodes to redeem fast enough. That's why we went towards the "print from SD" emulation trick, where Redeem reads the Gcode directly. The CPU is more than fast enough to keep up with the kinematics, it's just being starved for data really.

The M106 not being buffered as another G1 move is an issue, however, and should be worked on. I think it had to do with M106 bypassing the PRUs as it wasn't a synced movement, and the fan pins are controlled by the python-side instead of the PRU / path planner, @ThatWileyGuy or @eliasbakken can probably explain better.

ThatWileyGuy, 2018-01-17:
I'm taking a crack at the synchronization between move commands and non-move commands.

ThatWileyGuy, 2018-04-18:
This is currently fixed in staging.

closed 2018-04-18 by ThatWileyGuy