intelligent-agent/redeem #61: M109 doesn't wait for temperature to stabilize

jakep82, 2018-01-19:
When issuing an M109 at the start of a print, Redeem releases the wait as soon as the temperature reaches the set point.  The PID algorithm currently has issues, and I'm experiencing as much as 25 degrees of overshoot during warmup.  Because M109 releases the wait the instant the set temperature is reached, my prints start with the hotend more than 20 degrees above my set temperature.  I've been instructed on gcode hacks to get around this (extra M109 in start gcode with lower set temp), but it's not a proper solution.

While I understand work is happening to reduce the PID overshoot, M109 shouldn't release the wait until the temperature remains within a sane range of the set value (+/- 2 degrees?) for a reasonable length of time (5 seconds?)

darylbond, 2018-01-22:
``Heater`` already has a function available ``is_temperature_stable(seconds=10)`` that does this check, do we simply want to replace the ``is_target_temperature_reached()`` call in ``M116`` with this function?

Wackerbarth, 2018-01-22:
_stable sounds good to me. The only other use of temperature that I see is "at or above" to lock out "cold extrusion" and "at or below" to stop monitoring when shutting down. Both of these can be handled outside of the Heater object.

darylbond, 2018-01-23:
For how long should we require the temperature to be stable? Default for `is_temperature_stable` is 10 sec but I feel something like 3 sec is more appropriate.

jakep82, 2018-01-23:
From my point of view, something in the range of 3-5 seconds seems appropriate.