intelligent-agent/redeem #64: [develop] Revised Parser failing on M574

Wackerbarth, 2018-01-23:
Executing ```M574 Z2 z_neg, h_neg```
```
Jan 22 22:34:40 kamikaze redeem[2612]: 01-22 22:34 root         ERROR    Error while executing M574: 'CascadingConfigParser' object has no attribute 'set'
Jan 22 22:34:40 kamikaze redeem[2612]: 01-22 22:34 root         ERROR    Traceback (most recent call last):
Jan 22 22:34:40 kamikaze redeem[2612]:   File "/usr/local/lib/python2.7/dist-packages/Redeem-2.0.5-py2.7-linux-armv7l.egg/redeem/GCodeProcessor.py", line 127, in execute
Jan 22 22:34:40 kamikaze redeem[2612]:     self.gcodes[val].execute(gcode)
Jan 22 22:34:40 kamikaze redeem[2612]:   File "/usr/local/lib/python2.7/dist-packages/Redeem-2.0.5-py2.7-linux-armv7l.egg/redeem/gcodes/M574.py", line 36, in execute
Jan 22 22:34:40 kamikaze redeem[2612]:     self.printer.config.set('Endstops', 'end_stop_'+es+'_stops', config)
Jan 22 22:34:40 kamikaze redeem[2612]: AttributeError: 'CascadingConfigParser' object has no attribute 'set'
```

darylbond, 2018-01-23:
the new [config parser](http://configobj.readthedocs.io/en/latest/configobj.html) uses dictionary type accessing, so we should simply be able to change it to

`self.printer.config['Endstops']['end_stop_'+es+'_stops'] = config`

darylbond, 2018-01-23:
There were a few more issues that needed fixing as well, but I think I have this fixed in #66 .