intelligent-agent/redeem #68: Prints are frequently failing with Temperature falling too quickly alarm

pistolero, 2018-02-16:
```
Feb 16 09:50:12 kamikaze redeem[438]: 02-16 09:50 root         DEBUG    Poking watchdog
Feb 16 09:50:26 kamikaze redeem[438]: 02-16 09:50 root         WARNING  Disabling heaters
Feb 16 09:50:26 kamikaze redeem[438]: 02-16 09:50 root         ERROR    Alarm: Temperature falling too quickly (-131.710289175 degrees) for E
Feb 16 09:50:26 kamikaze redeem[438]: 02-16 09:50 root         DEBUG    Alarm executed
Feb 16 09:50:42 kamikaze redeem[438]: 02-16 09:50 root         DEBUG    Poking watchdog
```

This is probably caused by some electrical noise. Maybe it would be a good idea to add low-pass filter before that check.

Wackerbarth, 2018-02-16:
@pistolero -- Which version of redeem?

pistolero, 2018-02-16:
@Wackerbarth currently I'm experiencing this on staging.

ThatWileyGuy, 2018-02-16:
The hardware already has a low-pass filter - it's very likely that your thermistor is actually disconnecting momentarily and the low-pass filter is smoothing it so the reported temperature doesn't fall all the way to zero.

pistolero, 2018-02-16:
@ThatWileyGuy good to know.

It could also go up "Temperature is rising too quickly", especially if nozzle is bumping into bed at the beginning of the print. I had to set `max_rise_temp_e = 40.0` to avoid raising alarm at almost every print.

goeland86, 2019-01-16:
@pistolero as @ThatWileyGuy says - I would strongly recommend checking the stability of your thermistor wiring. The fact that you get an alarm when the nozzle bumps the bed indicates you likely have a crimping problem on your thermistor connection somewhere. I had to raise the value for `max_rise_temp_e` to 20 when I changed to a 40W cartridge instead of 30W, but that behavior is consistent and only when heating. Wiggling the wires does not affect my temperature readings at all.