intelligent-agent/redeem #69: Off topic please be nice :-)

nirgivon, 2018-02-17:
Hi,

I have been learning Linux on my one and trying to use the PRU's on the Beaglebone with out much success.
There is really hardly any good documentation and the process is different for different kernel releases making the little documentation out there even less effective.

I would really really really appreciate and be great-full if you can point me in the right direction where can I get information on how to successfully use the PRU's.

P.S
I use Replicate on my printer :-)

ThatWileyGuy, 2018-02-17:
What sort of pointer you want depends on how you want to use them.

What makes the documentation and examples so confusing is that there are two ways to communicate with the PRUs. There's pure shared memory, which is generally referred to as UIO and implemented by the kernels that end in -bone, and there's a message-passing framework called RPMsg implemented on top of a "remote processor" framework called RProc implemented by the kernels that end in -ti. RPMsg is definitely the easier of the two if you're starting out, but you'll find that most existing projects (Redeem/Replicape included) are using UIO because RPMsg/RProc weren't ready until a couple years after the BBB launched.

The other question is what platform you want to write code on. You do have a choice here - TI provides Code Composer Studio (CCS) and compilers for the PRU on Windows/Mac/Linux, so you don't necessarily have to code on your BBB (although you do have to copy the finished binaries to your BBB to run them, obviously).

This is a good place to start if you want RProc/RPMsg: http://processors.wiki.ti.com/index.php/PRU_Training:_Hands-on_Labs

Note that if you're on Windows, you just need CCS. You can install the PRU Code Generation Tools from within CCS, and you don't need the Linux Processor SDK at all. I haven't tackled this on Linux yet, but I can tell you that current Replicape Umikaze images come with the compilers and PRU Code Generation Tools pre-installed because we use them ourselves.

nirgivon, 2018-02-18:
Thanks so much for the detailed reply!

One last Q, if I want to dig more in to UIO, where can I find tutorials/books/explenations?
I have access to linux and dont  mind developing on the BBB.

Thanks again!

carlsoar, 2018-02-23:
@nirgivon Derek Molloy’s book seems to pop up a lot, it’s 2-3 years old now, but still relevant (I have it in my personal library), I suggest taking a look at that: http://exploringbeaglebone.com/chapter13/

Would it be ok to close this issue? Feel free to chat on the replicape slack :)

closed 2018-02-23 by nirgivon