intelligent-agent/redeem #72: Provide configuration option to invoke a script after M500 execution

cwhinton, 2018-02-19:
As a user I would like to be able to have redeem invoke a script I provide after it executes M500, saving the local configuration file. I would expect it to be called regardless of if I explicitly invoked M500 or if another process or macro does it on my behalf.

I would use this feature to submit the local.cfg to git, pushing them to the server immediately, so as to have a level of protection from catastrophic failure of the media, file system and stupid user tricks.

carlsoar, 2018-02-23:
@cwhinton, I think this would be doable... my personal thought is m500 should be called by the user, I don’t like stuff being overwritten unless I make adjustments, check them, then save when I’m happy with it.

I could add an optional flag e.g. M500 S1 which runs a bash script located in the same directory as the config, and where m500 s0 is default (no script). 

Thoughts?

carlsoar, 2018-02-23:
@goeland86 enhancement tag. :)

cwhinton, 2018-02-23:
I like it being optional.  I was envisioning a configuration entry in local.cfg to specify the script name that would default to blank.

If I recall properly the G29 macro that gets generated includes a M500. I can’t think of any other commands that also end up calling M500.

cwhinton, 2018-03-03:
I found a thing called incrontab that is designed to execute scripts upon file/directory change.  I'm going to look at implementing that rather the expect Redeem to be muddied up with this function.

Here's walk-through on using it.  https://www.howtoforge.com/tutorial/trigger-commands-on-file-or-directory-changes-with-incron/

I'll leave it to you guys to determine if you want to go forward with the enhancement.

cwhinton, 2018-03-06:
I'm going to go ahead and withdraw this as it's really would only be a partial solution since it wouldn't capture changes made outside of the redeem manipulating the configuration.

closed 2018-03-06 by cwhinton