intelligent-agent/redeem #81: Incorrect Motion Calculations with Slaved Axes

Wackerbarth, 2018-03-18:
Even though a pair of axes are slaved together, the planner calculates as if they are orthogonal.  In my use, I have Z and H for the steppers on a Z axis. The motion moves the correct number of steps on each stepper. But, as I discovered when attempting to probe, a 100mm distance is computed as if it is 141.4mm. In particular, because of the way the motion is interpolated, if the motion is interrupted by a probe, it concludes that the probe is much lower than its actual position.  Probing a single point from different heights gives very different results.  And, the speeds are incorrect.

ThatWileyGuy, 2019-01-27:
I believe this was fixed for RC in 4feba7b80a07b233cd935da310834d280f5ece21. I do not have a dual-Z-axis printer to confirm, though.

Wackerbarth, 2019-01-28:
The present code does handle totally "slaved" axes where 4 steppers operate as 3 orthogonal axes.. However, it does not handle motion which is a composite of multiple axes. In my use case, I want to define 4 axes which can move independently to determine the position of the nozzle. The present calculation assumes that all of the axes are orthogonal. But they are not. Thus it does not compute distances properly.

ThatWileyGuy, 2019-01-28:
I don't really understand the request. How are you setting up independent but not orthogonal axes?

Wackerbarth, 2019-01-28:
For one simple situation,
Think of the CORE-XY style calculation.

Using X, Y, Z to represent the cartesian nozzle position, and
using x, y, z1, z2 to represent the actuators.

∂X = ∂x
∂Y = ∂y
∂Z = (∂z1 + ∂z2)/2
∂ß = (∂z1 - ∂z2)/2

The ß axis might be used to adjust the relative alignment of two Z towers.

In this case z1 and z2, are co-linear.

But, in general, you have extendable tools. The extension of the tool could be at any angle relative to X, Y, Z.

ThatWileyGuy, 2019-01-28:
CoreXY is a supported motion platform by itself. If raw-motor movements need to be done that bypass the path planner's motion platform calculations, we support switching to cartesian mode and back.

I'm going to close this as the original issue is fixed.

closed 2019-01-28 by ThatWileyGuy

Wackerbarth, 2019-01-28:
Yes, in your rush to close issues, you can just "define away" the problem if you wish. But you are also refusing to recognize that there are other valid uses, for example, Core-XZ or multi-axis CNC. And you are, thereby "defining away" the usefulness of Redeem as a platform. I suggest that the issue remain to be addressed in a later release.

ThatWileyGuy, 2019-01-28:
This issue is titled "Incorrect Motion Calculations with Slaved Axes." That issue was described in detail, the causes were enumerated, and a fix was put in place to resolve it.

If you have an actionable issue or request, feel free to open another issue to track it. Otherwise, please keep these issues on-topic.