intelligent-agent/redeem #83: Comments get stripped from local.cfg upon write from Redeem

cwhinton, 2018-04-08:
Any action that causes local.cfg to be written from Redeem strips any comments that are present.

Commands that I've found so far that do this:
G31
G32
M500

Wackerbarth, 2018-04-08:
Please indicate the version of Redeem that you areusing.

cwhinton, 2018-04-08:
It happened with Umikaze-2.1.1 and continues to happen with Umikaze-2.1.2-rc5.

Wackerbarth, 2018-04-08:
G31 & G32 do not, in themselves rewrite local.cfg. They are macros and I suspect that you are calling some M-code from them.

cwhinton, 2018-04-08:
M574 as in
```
g32 = M574 Z2 z_neg,h_neg
g31 = M574 Z2
```

Issuing the G32 or G31 outside the context of G29 triggers the write, so it's not just the M500 within the G29.

example:
Place comments as follows in local.cfg then perform one of the actions that trigger save and the # comments are stripped.
```
[System]
machine_type = Hypercube Evolution
loglevel = 20
plugins = UserInterrupt
# working config after glass added.
# todo: fix probe offset
```

Wackerbarth, 2018-04-08:
Not disputing that it happens. But the fault is cause by M574 and only indirectly relates to G31 or G32.