intelligent-agent/redeem #91: Redeem 2.1.2 RC5c fails to start if servos are enabled

troy-jacobson, 2018-04-19:
When a servo is configured in 2.1.2rc5c, the adafruil pwm code throws and exception and stops the redeem process during initialization.  I'll post some logs as soon as i have a chance to revert my system to its original state.

troy-jacobson, 2018-04-19:
The simple solution is to revert to the non-adafruit pwm code.  If there is a need for both the new and old way of doing things, I have a simple patch ready.

ThatWileyGuy, 2018-04-19:
We should probably just revert to the non-Adafruit code. The file paths to the PWM devices change on >4.4 kernels, which is why the Adafruit code was integrated, but if we need both it's probably simpler to just modify our own code to detect which file path it should be using.

Wackerbarth, 2018-04-23:
Fixed in 2.1 at `b2d2f4d`

closed 2018-04-23 by Wackerbarth