intelligent-agent/redeem #96: Redeem doesn't recover from cancelling a print

cwhinton, 2018-04-23:
Using the 2.1.x branch.  Cancelling a print causes Redeem to consume ~94% cpu and close connection to OctoPrint.  OctoPrint is unable to reestablish the connection.  Steppers are powered off but the heaters continue to activate, indicated by lights on the replicape and temperature measurements taken with IR thermometer.  I've let the printer set for 10 minutes in this state without Redeem recovering.  Rectified by restarting Redeem with systemctl.  

I've triggered the behavior with both the 'normal' cancel button as well as OctoPrint addins that issue M112.

goeland86, 2018-04-23:
@cwhinton stupid question - do you have a similar issue if you use print from SD and use the cancel button then? I think it issues `M25` instead of `M112` in that case.

cwhinton, 2018-04-23:
I've not attempted any printing from SD.

Szeker, 2018-04-27:
@goeland86 - do you refer to the "emulated" printing from an SD card - introduced in 2.1.2 ?(Link: http://wiki.thing-printer.com/index.php?title=Print_from_sd)

If so, than I have the same experience with "printing form SD": after cancelling the print I have to restart and reconnect to Redeem (or even Octoprint, I'm not sure) to make it running again.

As I do not have all conditions in my mind - at which exact conditions the issue happened - I will re-check it and give feedback on it here.

goeland86, 2018-04-27:
@Szeker yes I am. Hrm. This is frustrating. I feel almost lucky that the release isn't done yet as it gives us time to fix it.

BTW any idea if there's a correlation with the OctoPrint settings? (please don't reply on that particular problem, this is just to know if you've modified the OctoPrint settings or not) https://github.com/foosel/OctoPrint/issues/2575

cwhinton, 2018-04-27:
I've not adjusted my OctoPrint settings but my M114 response looks like
```
Send: M114
Recv: ok C: X:5.0 Y:5.0 Z:29.3 E:-1.0 A:0.0 B:0.0 C:0.0 H:0.0
```
so should be parsed by OctoPrint according to the notes in the linked issue.

Wackerbarth, 2018-04-27:
Well, we made a change to put X,Y,Z toward the front.  My question has to do with the `C: ` just after the `ok`.
I don't think that it should be there.

troy-jacobson, 2018-04-29:
@cwhinton Are you using M574 command anywhere in your print?