intelligent-agent/redeem #103: Provide support for using sensors like FSR, Piezo, BLTouch for homing

cwhinton, 2018-04-29:
These sensors all reset shortly after being triggered.  Well at least the JohnSL board for Force Sensing Resistors and the Precision Piezo board do.  The Wiki indicates the BLTouch does as well.

Wackerbarth, 2018-11-08:
Testing needed -- @ThatWileyGuy has included a "fix" for this condition in #138. Would those who have these sensors please test that code and report if it resolves their problem?

goeland86, 2018-11-08:
@Wackerbarth afaik the fix is **not** for homing, only probing.

There is apparently a clear misunderstanding in that a "pulse" z-probe in redeem cannot yet be used for Z homing - but this has always been the case and should not delay release of a fix.

As to testing, I've run @ThatWileyGuy 's branch in #138 on both FSR and piezo sensors on my delta and it works probes fine.

goeland86, 2019-01-27:
@Wackerbarth can you check if @ThatWileyGuy 's latest fixes for pulse-probing fixed the z-probe homing? Private conversation suggests it should

Wackerbarth, 2019-01-28:
I don't have the hardware to test this. However, I also think that the latest code should work.

goeland86, 2019-03-31:
I can confirm that on my piezo build this has been fixed in `RC`. @cwhinton do you concur?

cwhinton, 2019-03-31:
@goeland86 I'll have to spend some time re-establishing state on my printer.  I've not done much with it in the past year.

cwhinton, 2019-04-09:
As it turns out, when I tried to integrate the Precision Piezo with the Flex3Drive, I managed to remove one of the wires from the piezo and the combination of the very thick aluminum bed in combination with the glass sheet is too heavy for the FSR.  I cannot test this scenario any longer.

goeland86, 2019-04-09:
@cwhinton well that's unfortunate... Are you using piezos or FSR? The piezos should work as long as the loading doesn't exceed the parameter (I've got a bit 350mm aluminum + glass bed on them), FSRs will lose precision and therefore be less precise the higher the base loading on it.

zaped212, 2019-08-26:
@goeland86,

Not sure if it helps but, I have my printer set based off a commit after #138  was merged and things seem to be working properly for me.

Never noticed any issue with my piezos trigger ( Precision Piezo inline with hotend ) not being detected by redeem other than when my piezo sensitivity was too low.

goeland86, 2019-08-26:
@zaped212 yeah I've not had a problem with my piezos either. not in a long time.
I have a different issue with G33 on a delta I need to properly document, but the probing seems to work.