intelligent-agent/redeem #105: Custom Temperature Sensors

mloit, 2018-04-30:
Currently adding a custom temperature sensor is a real pain. It would be great if TemperatureSensorConfigs.py could be relocated to the redeem working directory `/etc/redeem`. Or even better, some mechanism added into the config file to be able to define the sensor.

troy-jacobson, 2018-05-02:
How much customization do you mean?  Adding a new thermistor table, direct connect to the BBB's adc, external ADC over i2c, something else?

mloit, 2018-05-02:
Just the ability to add new entries to the tables that already exist in an easier way.  Could be as simple as relocating `TemperatureSensorConfigs.py` to a more accessible location than `/usr/local/lib/python2.7/dist-packages/Redeem-[VERSION]-py2.7-linux-armv7l.egg/redeem` perhaps placing it in `/etc/redeem`?  Or my preference would be to have it on the config file in the `[Heaters]` section, providing a mechanism to add the parameters for the sensor, instead of directly editing the table... perhaps something like: 
```
sensor_E = CUSTOM(thermistor, "Semitec-104NT4", 4700, 0.797110609710217e-3, 2.13433144381270e-4, 0.65338987554e-7)

sensor_E = CUSTOM(pt100, "E3D-PT100-AMPLIFIER", 4700, 100, 0.0039083, -5.77e-07)

sensor_E = CUSTOM(tboard, "Thermocouple", 0.005)
```