intelligent-agent/redeem #107: Redeem Crashes when invalid M-code issued

mloit, 2018-04-30:
While testing for support of M0 [and other program control codes, M1, M2, M30], redeem became unresponsive.  Issuing M0 alone has no effect, but issuing M00 crashes the system. Same goes for M01, M02. [M2 Also crashes, but M1 does not], M30 also crashes. [it actually looks like M0 and M1 are never sent, as they never come up in the OctoPrint terminal window] Invalid codes should be handled in a safe/recoverable manner.

```Apr 30 04:12:14 kamikaze redeem[426]: Exception in thread Pipe:
Apr 30 04:12:14 kamikaze redeem[426]: Traceback (most recent call last):
Apr 30 04:12:14 kamikaze redeem[426]:   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
Apr 30 04:12:14 kamikaze redeem[426]:     self.run()
Apr 30 04:12:14 kamikaze redeem[426]:   File "/usr/lib/python2.7/threading.py", line 754, in run
Apr 30 04:12:14 kamikaze redeem[426]:     self.__target(*self.__args, **self.__kwargs)
Apr 30 04:12:14 kamikaze redeem[426]:   File "/usr/local/lib/python2.7/dist-packages/Redeem-2.1.2-py2.7-linux-armv7l.egg/redeem/Pipe.py", line 89, in get_message
Apr 30 04:12:14 kamikaze redeem[426]:     self.printer.processor.enqueue(g)
Apr 30 04:12:14 kamikaze redeem[426]:   File "/usr/local/lib/python2.7/dist-packages/Redeem-2.1.2-py2.7-linux-armv7l.egg/redeem/GCodeProcessor.py", line 129, in enqueue
Apr 30 04:12:14 kamikaze redeem[426]:     self.resolve(gcode)
Apr 30 04:12:14 kamikaze redeem[426]:   File "/usr/local/lib/python2.7/dist-packages/Redeem-2.1.2-py2.7-linux-armv7l.egg/redeem/GCodeProcessor.py", line 94, in resolve
Apr 30 04:12:14 kamikaze redeem[426]:     gcode.command = self.gcodes[gcode.code()]
Apr 30 04:12:14 kamikaze redeem[426]: KeyError: 'M00'
```

carlsoar, 2018-05-02:
Just to add, I typed G303 accidentally and that crashed it too, so not just M Codes.

zaped212, 2019-01-26:
Looking at the latest version of the RC branch it looks like this has already been resolved.
The code now checks to see if the requested gcode "key" currently exists before trying to access it.

Wackerbarth, 2019-01-26:
You are correct.

closed 2019-01-26 by Wackerbarth