intelligent-agent/redeem #140: local.cfg

think-inhell, 2018-11-11:
Make changes in local.cfg with `nano` and octoprint
After making changes in local.cfg and restarting redeem service:

```
root@kamikaze:/etc/redeem# systemctl status redeem
● redeem.service - The Replicape Daemon
   Loaded: loaded (/lib/systemd/system/redeem.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-11-11 09:16:42 UTC; 37s ago
  Process: 1466 ExecStart=/usr/local/bin/redeem (code=exited, status=1/FAILURE)
 Main PID: 1466 (code=exited, status=1/FAILURE)

Nov 11 09:16:42 kamikaze redeem[1466]:   File "/usr/lib/python2.7/ConfigParser.py", line 324, in readfp
Nov 11 09:16:42 kamikaze redeem[1466]:     self._read(fp, filename)
Nov 11 09:16:42 kamikaze redeem[1466]:   File "/usr/lib/python2.7/ConfigParser.py", line 512, in _read
Nov 11 09:16:42 kamikaze redeem[1466]:     raise MissingSectionHeaderError(fpname, lineno, line)
Nov 11 09:16:42 kamikaze redeem[1466]: ConfigParser.MissingSectionHeaderError: File contains no section headers.
Nov 11 09:16:42 kamikaze redeem[1466]: file: /etc/redeem/local.cfg, line: 1
Nov 11 09:16:42 kamikaze redeem[1466]: 'direction_x = -1\n'
Nov 11 09:16:42 kamikaze systemd[1]: redeem.service: Main process exited, code=exited, status=1/FAILURE
Nov 11 09:16:42 kamikaze systemd[1]: redeem.service: Unit entered failed state.
Nov 11 09:16:42 kamikaze systemd[1]: redeem.service: Failed with result 'exit-code'.

cat /proc/version
Linux version 4.1.38-bone24 (root@a2-imx6q-wandboard-2gb) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.4) ) #1 Fri Jan 20 13:05:02 UTC 2017

cat /etc/kamikaze-release
Kamikaze 2.1.0 28-10-2018

```

goeland86, 2018-11-11:
@think-inhell can you share what your local.cfg is?

It looks like you didn't specify the section headers in local.cfg.

The default.cfg file has sections for each value, like
```
[Geometry]
# 0 - Cartesian
# 1 - H-belt
# 2 - Core XY
# 3 - Delta
axis_config = 0
...
[Delta]
# Length of the rod
L   = 0.322
# Radius of the columns
r   = 0.175
...
```

And what the error message says is that it can't find the same sections in your `local.cfg`.

think-inhell, 2018-11-11:
@goeland86 Thanks for your fast reply

My current `local.cfg`: (everything commented out)
```

#direction_x = -1
#direction_y = -1
#direction_z = -1

#steps_pr_mm_x = 6.25
#steps_pr_mm_y = 6.25
#steps_pr_mm_z = 6.25

#invert_X1 = False
#invert_Y1 = False
#invert_Z1 = False
#invert_X2 = True
#invert_Y2 = True
#invert_Z2 = False
```
So I've put everything in my new `kossel_try1.cfg`

```
#config Testfile for MiniKossel
[System]

machine_type = Kossel_mini_try

[Geometry]
# Delta
axis_config = 3

# Set the total length each axis can travel 
travel_x = -0.5
travel_y = -0.5
travel_z = -0.5

# Define the origin in relation to the endstops
offset_x = -0.05
offset_y = -0.05
offset_z = -0.05

[Delta]
# Distance head extends below the effector.
Hez = 0.055
# Length of the rod
l = 0.2200
# Radius of the columns
r = 0.1532

# Effector offset
Ae  = 0.047
Be  = 0.047
Ce  = 0.047

# Stepper e is ext 1, h is ext 2
[Steppers]
current_x = 0.4
current_y = 0.4
current_z = 0.4
current_e = 0.4

steps_pr_mm_x = 6.25
steps_pr_mm_y = 6.25
steps_pr_mm_z = 6.25
steps_pr_mm_e = 6.0

# Which steppers are enabled
in_use_x = True
in_use_y = True
in_use_z = True
in_use_e = True

direction_x = -1
direction_y = -1
direction_z = -1

slow_decay_x = 1
slow_decay_y = 1
slow_decay_z = 1
slow_decay_e = 1
slow_decay_h = 1

microstepping_x = 6
microstepping_y = 6
microstepping_z = 6
microstepping_e = 6

[Endstops]
end_stop_X1_stops = x_cw
end_stop_Y1_stops = y_cw
end_stop_Z1_stops = z_cw
end_stop_z2_stops = x_ccw, y_ccw, z_ccw

soft_end_stop_min_x = -0.05
soft_end_stop_min_y = -0.05
soft_end_stop_min_z = -0.001

soft_end_stop_max_x = 0.05
soft_end_stop_max_y = 0.05
soft_end_stop_max_z = 0.3

has_x = True
has_y = True
has_z = True

# Invert =
#   True means endstop is connected as Normally Open (NO) or not connected
#   False means endstop is connected as Normally Closed (NC)
invert_X1 = False
invert_Y1 = False
invert_Z1 = False
invert_X2 = True
invert_Y2 = True
invert_Z2 = False

[Homing]
home_x = 0
home_y = 0
home_z = 0
home_speed_x = 0.2
home_speed_y = 0.2 
home_speed_z = 0.2
home_speed_e = 0.2
home_speed_h = 0.2 

[Cold-ends]
connect-therm-E-fan-1 = True
add-fan-0-to-M106 = True
add-fan-3-to-M106 = True

[Planner]
# Max speed for the steppers in m/s
max_speed_x = 0.4
max_speed_y = 0.4
max_speed_z = 0.4
max_speed_e = 0.4
max_speed_h = 0.4

[Probe]
offset_x = 0.0
offset_y = 0.0
```

goeland86, 2018-11-11:
Right, ok.

What you're missing local.cfg is the section headers. It should be
```
[Steppers]
#direction_x = -1
#direction_y = -1
#direction_z = -1

#steps_pr_mm_x = 6.25
#steps_pr_mm_y = 6.25
#steps_pr_mm_z = 6.25

[Endstops]
#invert_X1 = False
#invert_Y1 = False
#invert_Z1 = False
#invert_X2 = True
#invert_Y2 = True
#invert_Z2 = False
```

think-inhell, 2018-11-11:
Thanks :) My fault. With the headers it worked perfect.

closed 2018-11-11 by think-inhell