intelligent-agent/redeem #141: PathPlanner/PRU hangup during brim

PaoloBi, 2018-11-14:
Hi all, I am trying to print a large object (390x46x101mm) that needs brim (long and slow moves). PathPlanner/PRU hang-up always at line 312, no more G1 codes accepted. M105 requests are answered OK, so I suppose there's some problem in PRU. 
Perhaps this issue has already solved ? I see that a new path planner has been released, shall I try that ?
I attach my .gcode
[TestPB_V1.gcode.txt](https://github.com/intelligent-agent/redeem/files/2579641/TestPB_V1.gcode.txt)
Thank you
Paolo

PaoloBi, 2018-11-14:
I stripped down my gcode file, now hangup occurs at line 38. It happens every time. weird....Here it is:

[test2.gcode.txt](https://github.com/intelligent-agent/redeem/files/2579926/test2.gcode.txt)

Wackerbarth, 2018-11-14:
1) You give no hint as to what version of Redeem you are running. Yes, you really should try this with the "RC" branch. That has the latest path code.
2) Have you looked in the journalctl log to see it there are any messages that might indicate a problem?
3) It would also be helpful to know your configuration.

PaoloBi, 2018-11-14:
Hi, yes you're right. My version is 1.2.7. 
Quite old, but as I added many custom modules (PT1000, I2C I/O and sensors, autolevel etc) I did not upgrade my version since a long time. I never modified the planner/PRU code, anyway. 
I did not find any special message in journalctl log.
My setup is a modified Replicape with a cartesian hardware, controlled by Octoprint. I enclose a communication log, as you can
[image]
 see M105 are answered while G1 command are ignored. I don't think this is a "ok sync" problem, as execution stops always exactly at the same point....

PaoloBi, 2018-11-15:
Update: I tried the "RC" branch. I copied the path_planner folder over mine, re-compiled all with python setup.py install, restarted service, printed same file....no relief. 
May someone test my file ? It is just a 90-seconds job...
thanks

goeland86, 2018-11-15:
@PaoloBi how did you install the `RC` branch? If you were running Redeem 1.2.7, I guess this means you had installed Kamikaze 2.1.0?

If so you'll need to install the newer 2.2.1-dev4 image and pull the `RC` branch from that. There are significant updates to the kernel interactions and the likelihood that you have a mix of Redeem 1.2.7 and RC in `/usr/local/lib/python2.7/dist-packages` is high.

I'll try to run your gcode file tonight to validate that the latest RC branch runs it fine.

PaoloBi, 2018-11-15:
Thank you for helping, in the meanwhile I reverter to old pathPlanner and logged while printing, I enclose it for  troubleshooting

[PathPlannerLog1.txt](https://github.com/intelligent-agent/redeem/files/2586130/PathPlannerLog1.txt)

PaoloBi, 2018-12-10:
I found another file that hangs the path planner **almost immediately** (first move). I commented out the warmup commands, so this file can be checked out in few seconds. 
Yesterday I downloaded the latest master branch of redeem, and the path planner is identical to mine, so as I can see we have 3 possibilities:
1. my test file is OK when printed on another BBB/printer, an I am doing something terribly wrong (hopefully);
2. my test file will hang "official" path planner, so this should be fixed !
3. the master branch path planner is obsolete and we MUST migrate to RC version, in this case I would like to be sure before starting my migration

Again, I enclose a test file for the willing

[flangia1.zip](https://github.com/intelligent-agent/redeem/files/2663716/flangia1.zip)

goeland86, 2018-12-10:
@PaoloBi thanks for the update.

I understand this will cause a freeze of the planner from the `master` branch. However all development is now focused on the `2.2.x` base, with `RC` receiving the latest patches that will be included in an upcoming image release. Unfortunately if you were running Umikaze 2.1.1 or earlier, you'll need to update the complete Umikaze image, as the `2.2.x` branch was made specifically to run on new kernel versions than what is available in previous Umikaze images.

There's a `221dev4` RC Umikaze image available in the release page that you can try if you're in a hurry, otherwise we'll issue an official RC image probably in the next few weeks.

PaoloBi, 2018-12-15:
After a message from ThatWileyGuy I gave a try to path_planner from **develop** branch and deadlock has gone away, while it seems to continue in **master** branch. I will check this more carefully in next hours.

Wackerbarth, 2018-12-15:
@PaoloBi -- As stated above, ALL development is on the `RC` branch which is based on the `2.2.x` branch. `master` and `develop` are known to have problems and those branches are deprecated. Unless you want to fix something for yourself, you will need to use the branches that we recommend and are actively supporting.