intelligent-agent/redeem #142: M27 replies inconsistent for OctoPrint

goeland86, 2018-11-26:
See https://github.com/foosel/OctoPrint/issues/2910 for the problem.
Seems that Redeem is sending current print byte and then the `Not printing from SD`.

Bonus: we now have a spec to follow for SD-card printing methods.

Wackerbarth, 2018-11-27:
Trying to understand how the system is supposed to behave with respect to printing from a file.. Is this description correct?
The SD Unit is a "black box" that provides a collection of available selections and commands to maintain that collection. It also provides a "portal" which supplies the contents of a selection. The M27 command provides the status of the portal. Initially, the portal is "detached". (M27 == "Not SD printing.").  M23 "attaches" the portal to the designated selection (M27 == SD printing byte 0/235422). M24 starts printing (M27 == SD printing byte 1000/235422). When the print finishes, the portal returns to the "detached" state (again, M27 == "Not SD printing.").

goeland86, 2018-11-27:
@Wackerbarth the current code is basically jumbling its responses to `M27` queries, sending back 2 status responses instead of a single one (`SD printing byte x/xyz` then `Not SD printing` instead of an XOR on the two).

I can fix this if it's not clear how SD printing was meant to work. In any case this will be moot when OctoPrint 1.4.x comes out and we have our own communication layer between Redeem & OctoPrint.

goeland86, 2018-11-27:
My understanding is that Marlin (which OctoPrint sort of expects, G-Code behavior-wise) does the latter, where a repeat print requires another M23.

The idea being that you can do M25 to pause a print from SD, interact outside the SD-file through normal GCode (i.e. filament switch, etc.), then M24 to resume the print.

Wackerbarth, 2018-11-27:
I agree that "pause" leaves the file connected (at its present byte position). The question only concerns what happens once we reach the end. After cleaning the bed, can I print another copy by simply sending another M24? That would be a reasonable behavior for a stand-alone printer with a "Print" button.

goeland86, 2018-11-27:
So, OctoPrint (either from the web or through Toggle endpoints) issues a new M23 before a new M24 for printing a new copy of the file.

I suggest that we then disconnect the file after the print completes

goeland86, 2018-12-17:
Further discussion is ongoing in #145, closing here.

closed 2018-12-17 by goeland86