intelligent-agent/redeem #147: M500 saves unchanged values

eliasbakken, 2018-12-12:
When running M500, the r and l are saved even if it is not changed. 
If steppers are disabled, the stepper current will be set to 0. 
ACs:
- [ ] When running M500, the "l" (small L) and "r" values are not saved to local.cfg
- [ ] When running M500, the stepper current is not changed to 0.

Wackerbarth, 2018-12-12:
There is a major replacement for the configuration which has been "waiting in the wings" for almost a year. Any changes to the current implementation will not be applicable to the new version.

goeland86, 2018-12-12:
@Wackerbarth until I see concrete implementations of that config change, the fixes @eliasbakken describes **need** to be addressed. I don't care if it clashes with the new config architecture, we'll deal with it when it comes.

Wackerbarth, 2018-12-12:
Well, I contend that any changes someone might propose should come AFTER the changes which are being made to handle Revolve, etc. And the only thing that has been holding those configuration changes up is that the path logic has not been stabilized in a releasable build. I have held them to help avoid the potential introduction of new issues which might further delay things.

If some new "contributor" submits changes to this issue, they are going to be VERY disappointed when their contribution winds up in the "bin" because another change has already addressed the issue.

goeland86, 2018-12-12:
@Wackerbarth the changes that are highlighted are bugs that should be addressed in the upcoming release. Your planned configuration changes are, to me at least, a while in the future.

And as such, a bugfix **now** is not ending up in the bin if the new functionality prevents the bugs listed above from re-occuring.