intelligent-agent/redeem #161: Probe offset does not seem to take affect

zaped212, 2019-01-11:
I previously was trying the umikaze release on my a4a board, but was having an issue after performing the g29 or g33 commands. I also tried updating to the tips of the RC branch as it seems to have the latest development.

I am using a Precision Piezo 20 inline probe ( hotend is the probe )

In my configuration I have:
[Probe]
offset_x = 0.0
offset_y = 0.0
offset_z = 0.00440563574938

After performing either the g29 or g33 commands I home my printer and then slowly move down to make sure I do not get a head crash by issuing:
g1 z10
g1 z6
g1 z3
g1 z2

My hotend always ends up triggering at 2.2mm even after performing a the leveling commands.

zaped212, 2019-01-16:
@goeland86 I presume you have a the bed leveling logic functional on your printer.
Can you tell me what command / set of commands you are using to accomplish that?

I was testing with G33 as that is what I was previously told to use, but always seemed odd to me compared to G29.

I have tried larger z offsets ( around 10mm ) but wasn't seeing a difference.

Currently I am just looking for any sort of change in hotend position and then I can actually start calibrating from there.

goeland86, 2019-01-16:
@zaped212 so yes, I have a probing setup on my delta that works. However my probe offset is very small as I'm using a piezo bed sensor, with the nozzle triggering it when it touches the bed.

However you can set your probe offset by using G34. Basically bring your nozzle down to height zero, then issue G34 - it will rise 5mm, "undock the probe" with G32, probe once and compute the offset, then "dock" the probe with G31. M500 saves the offset value in local.cfg.

After that I run G29.1 to generate a circular probe routine (with appropriate D and K parameters for bed diameter and probing speed). M500 saves the routine to local.cfg again.

Running G33 will execute the same probe pattern as G29 will, but it will instead compute delta factors (G33 N3 - endstop offsets only, G33 N4 - endstop offsets + delta radius, etc., doc here: http://wiki.thing-printer.com/index.php?title=Redeem#G33:_Autocalibrate_a_delta_printer - beware that Fn is no longer active on recent code, it's now Nn)

Hope this helps? If it still makes your Z height wrong, then we need to look at subtracting the Z probe offset from the endstop offsets I think, but I don't know if @ThatWileyGuy has that covered in the code yet or not.

zaped212, 2019-01-16:
@goeland86 
I am actually using the piezo hotend sensor :)

I will try again with the commands you have provided. I presume you are doing the actual bed leveling with G29 then?

Note: I have already created the probe pattern with G29.1

As for the Z offset of the probe I am currently making a change that will have the G30 probe command the probes z offset into affect: https://github.com/intelligent-agent/redeem/pull/169

That should help consolidate the probe related logic.

goeland86, 2019-01-16:
@zaped212 no - no G29 is ever called. G29 is only to be used if your bed is not planar (for a delta this means not situated flat mind you, just a flat surface, even if it's mounted crooked), and mostly helpful in the case of a cartesian/coreXY bed.

In a delta, by adjusting the endstop offsets and the delta radius, you usually adjust the plane of motion to the plane of the bed with the probing. If you're using a piezo sensor, I recommend using a K200 parameter for G29.1, otherwise you go too fast and lose precision as to the triggering timing. Too slow and it isn't quite reliable enough.

I do hope this will help, but I'm curious to see what @ThatWileyGuy has to say about needing to implement probe offsets within G30. I suspect he's taking the probe offset into account at another point in the code instead.

zaped212, 2019-01-16:
@goeland86
Yea I searched the codebase for the probe z offset and it was only found in the G33 command so I removed that usage to make it consistent.

And thanks, I will retry with the G33 command and see how that goes.

goeland86, 2019-01-16:
@zaped212 if you have more questions, feel free to join us on the Replicape Slack channel, the self-invitation link is on thing-printer here: https://www.thing-printer.com/wp-login.php?action=slack-invitation

zaped212, 2019-01-16:
Closing this issue to move to slack.
Will re-open if we find a problem.

closed 2019-01-16 by zaped212

zaped212, 2019-01-18:
Some closure on this problem:
The issue was being caused by the [Homing] home_z parameter being larger than the [Geometry] offset_ values.

This caused my printer to Home ( G28 ), then attempt to move up ( though the endstops were preventing that movement ) This caused the printer to think it was at Z = 530. and effectively cause it to ignore the applied endstop offsets. Once the max height was adjusted to 520 the offsets started working.

The Geometry offsets are around 526mm so having a home_z of 520 causes the hotend to move down slightly.

I can now issue G33 N4 P; G28, G1 Z0 and the hotend doesnt head crash, instead the hotend rests exactly on the bed where it is supposed to.