intelligent-agent/redeem #174: Update test code to reduce mock usage

zaped212, 2019-01-16:
While working on updating the tests for G30 and G33 for changes I was making that would break the test code. I discovered there were several parts of the existing code that were overly mocked and prevent testing of the full feature.

G33 for example is unable to test when the printer has a probe z offset applied since self.printer.processor.execute is being mocked and prevents execution of the G30 command which now applies the probe z offset.

Some of the items I saw being mocked:
- self.printer.path_planner.probe = mock.Mock(return_value=12.34 / 1000)
- self.printer.path_planner.get_current_pos = mock.Mock(return_value={ "X": 10.0, "Y": 20.0, "Z": 35.0 })
- self.printer.config.getfloat = mock.Mock(return_value=11.22)
- self.printer.processor.execute = mock.Mock()

These were causing unexpected results ( before I noticed them ) specifically getfloat being mocked as that was currently being used to retreive the probe x and y offset, and I was adding the z offset.

One of the big problems is that anything trying to retreive a float value from the configuration file would now retreive a value of 11.22 Meters ( dont know about you but that is one heck of a lever arm for the probe ) and was causing my tests that apply the probes z offset to be over 11 meters off ( Now I would love a functional printer that was capable of over 11 meters of print height :) )

I propose, in most cases, we move the mocking down to a lower level, such that we mock as close to actual printer movements as we can.

For example:
self.printer.path_planner.probe could be mocked such that it gets the probe length and z_offset, and if the bed ( z0 ) is within the probe length with z_offset applied then we return that probed point, else like hardware we return the point where the hotend was at the end of the probe travel.

I also suggest we mock the printer movement, such that we could call G0, and have the printer position actually move as expected.

This would then require us to update the test code to then setup additional information ( such as printer position ), instead of just mocking that data.

Wackerbarth, 2019-01-16:
I see multiple levels of testing and mocking. At one level you might want to isolate a particular function. At a higher level, you want to test that functionality in the context of the system, and, at an even higher level, you might perform the test on actual hardware. I think we need to look at the overall organization of the way we define contexts so that we have good points to which we can apply mocking. The present structure is somewhat ad-hoc and unclean.

zaped212, 2019-01-16:
@Wackerbarth 
I completely agree that different levels are required depending on the specific test circumstances.

I just think it would be good to mock as much of the "hardware" level as we can by default ( create a virtual printer ), and then as tests require it they can mock higher levels if specifically needed instead of mocking higher levels by default.

One of the mocks that really "confused" me ( and took me a while to figure out what was happening ) was mocking the self.printer.config.getfloat function, as it would have been much cleaner to just set the required values in the config instead of mocking all of them.

Wackerbarth, 2019-01-16:
I think that I agree with you. However, when I think of “high" and “low”, I place mocking a virtual printer as a high level approach and mocking the return from a particular call as the low level version.

There is the operational question of whether you generate actual configuration files for a group of tests or just replace the whole configuration with a “constant” that is built into the test.

I've been trying to separate the configuration data values from the mechanism that loads them for just that reason.  I'm also trying to split off the parts that are extracted from the hardware into their own module for the same reason.

But I’ve been running into resistance because there is no immediate change in functionality and no one seems to understand the long term value in making such distinctions.

Richard Wackerbarth
rkw@dataplex.net

> On Jan 16, 2019, at 10:20 AM, zaped212 <notifications@github.com> wrote:
> 
> @Wackerbarth <https://github.com/Wackerbarth>
> I completely agree that different levels are required depending on the specific test circumstances.
> 
> I just think it would be good to mock as much of the "hardware" level as we can by default ( create a virtual printer ), and then as tests require it they can mock higher levels if specifically needed instead of mocking higher levels by default.
> 
> One of the mocks that really "confused" me ( and took me a while to figure out what was happening ) was mocking the self.printer.config.getfloat function, as it would have been much cleaner to just set the required values in the config instead of mocking all of them.
>

zaped212, 2019-01-16:
@Wackerbarth 
Yea I think we are talking about the same thing, just we have our personal view of high vs low flipped with eachother.

As for the configuration. I was thinking about just the default config, and then in the test setup, any test specific configurations can be applied.

Wackerbarth, 2019-01-16:
"flipped" -- Yes, but just like "left and "right" in theatre, (audience view or actors view), it only matters when it leads to confusion in the communication. 

But that still leaves open the question of how we can most conveniently transfer a particular set of configuration values into each specific test. For the bulk of the tests, do we create actual configuration files that get parsed? Or do we mock the configuration parser and directly supply the required values?

zaped212, 2019-01-16:
Again correct, we will need to figure out a naming convention to avoid confusion but we get the idea :)

I took a look at the mockprinter.py code and it looks like we are already providing the default config file along with providing stub versions of the local.cfg (my_test_local.cfg) and printer.cfg (my_test_printer.cfg). Both of those files set the configuration version, and the local.cfg file disables file logging.

The test classes are inheriting from mockprinter wich does create a full printer instance ( hooks up to redeem, mocks out some hardware functions )

Expanding the existing mockprinter class would probably be the best place to start mocking out additional hardware level functions eg probe.

Wackerbarth, 2019-01-16:
Rather than writing and causing custom configurations to be loader for a group of tests, I'm thinking that it might be easier to add some calls in the main program to a few "hook" functions that do nothing but provide an easy place for us to simulate particular configuration variables. (For example with a hook after the configuration has been loaded, we could set the variables for a particular test.)

Our testing infrastructure has been added "after the fact" and is still in its infancy. Improving that testing structure is a whole project. And that project is just as complex as the printer controller itself.

zaped212, 2019-01-17:
For the custom configurations I was actually thinking about just having the test SetUp functions call set to write to the configuration file. Its already doing that in some cases, and in others the read config functions are just being mocked there anyways so that would be an easy place for each test to control its own deviations from the default configuration file. Then we would let pytest handle calling the Setup function for us.

Note: Individual test functions will end up doing this anyways ( such as my changes to the G30 test function ) since different test functions ( for the same command ) could have opposing configuration values. This also re-inforces that having the Setup handle the generic changes for each of that units tests and then having each test supplement the changes would be pretty easy to maintain and does not have much complexity.

Wackerbarth, 2019-01-17:
Here's where my thinking is headed. Consider the configuration to be nothing more than a dictionary of printer control parameters by conveniently ignoring issues of cascading definitions and saving changes. Now we can use mock to replace the actual configuration building mechanism with a simple dictionary. 

Many of the tests don't really need even the default configuration. The MockPrinter class, or a SimpleMockPrinter class derived from it, would provide the dictionary and might seed it with some values.

Then the setup for the test class can add values that apply to the class. And, finally, the individual tests can make custom settings, all directly in the supplied dictionary.

This avoids the need to actually write, and subsequently clean up, a temporary configuration file.

For another group of tests, if we have moved the derivation of the stack of configuration file names into a function, rather than having it "inline" in the Redeem class, we can then mock that function to access static cases of the various configuration files.

Thus, rather than having any actual code directly embedded in it, the main Redeem class becomes little more than a list of calls to a collection of functions that perform the actual work. And each of those can be mocked out as appropriate.

Wackerbarth, 2019-01-17:
I'm not suggesting that it is "wrong", it's just that for testing at some level, you may want to avoid values from the default config without having to assure that, for the test, you found all of them and over-road/eliminated them. This is particularly necessary if you have a behavior that is "optional" and, to provide simplicity for the common case, the option is enabled in the default configuration.

Wackerbarth, 2019-01-17:
The in depth testing that you describe is desirable at some point in the release cycle. However it is not something that should be done by default. Most changes affect either every configuration, or only a limited number of them. From a development perspective, running the same test on multiple configurations that are highly likely to give the same result in each case is wasteful of the testing resource and that leads to an increased turn-around time.

zaped212, 2019-01-17:
@goeland86  I have to agree with @Wackerbarth  here, I dont see that we will be gaining much benefit from running an exhaustive list of configuration files on merge requests. Much of the code is not specific to any given type of printer or even printer configuration.

goeland86, 2019-01-17:
@zaped212 obviously the configuration test-matrix should only be done for those places where it makes a difference. But still, re-using the values from default.cfg would be easier and more sensible in the general case instead of mocking the printer.

What I'm aiming at is that we need a mechanism to validate that the values in our pre-defined config files are coherent and don't break the code in any way. It may not be necessary to run those tests on every PR or commit, but at a weekly interval to use as a "we didn't break things" report.

zaped212, 2019-01-17:
@Wackerbarth in response to your message: https://github.com/intelligent-agent/redeem/issues/174#issuecomment-455027748

I do like your idea of turning the configuration into a dictionary ( for testing ). I agree this would help prevent configuration pollution as the dictionary just lives in memory and could be easily purged / reset on each test.

Though, I would still argue that we would want to populate that dictionary with the values from the default configuration file. At a minimum this would give us sane starting defaults then the test could override them as it sees fit. I dont think overriding config values should be up to the mocked printer.

We already have capabilities of loading that configuration file so I see adding this dictionary as a simple extension of the existing functionality.

Wackerbarth, 2019-01-17:
The reason for running some tests without the default.cfg is to assure that those tests are not passing only because of some value that is in the default configuration file. Fortunately, once we get the configuration data isolated from the configuration file management, the difference in the formulation of the test setup is reduced to a single line.

goeland86, 2019-01-18:
> The reason for running some tests without the default.cfg is to assure that those tests are not passing only because of some value that is in the default configuration file. Fortunately, once we get [...]

I would argue that if you can set a value that makes it run, and another which makes it fail, you need to define an acceptable range for that value, test the range to make sure it passes, validates it breaks outside the range, then document that range in the wiki.

Wackerbarth, 2019-01-18:
The test that you describe is not the same as the "it does not depend on" test. And to exhaustively test in the manner that you describe, you would have to include tests for every possible combination of the parameters. So, having tests which never have a parameter defined is an important part of the overall test scheme.

goeland86, 2019-01-18:
> So, having tests which never have a parameter defined is an important part of the overall test scheme

But if you don't depend on parameters at all - why would you need a mock configuration in the first place?

Wackerbarth, 2019-01-18:
It's not that you don't depend on ANY parameters. Rather it is that there are SOME parameters that you wish to eliminate for the test. So you still need the mock.

Wackerbarth, 2019-01-18:
The standard procedure in testing is to collect a group of tests that run in a similar environment and provide one setup for that group of tests. Some tests run in the scope of the entire printer. Other tests are limited to a particular subsection. Both kinds of tests likely depend on SOME configuration parameters. Having a single insertion point for the configuration parameter insertion makes sense. But that does not imply that it should always utilize the full configuration mechanism to produce the desired environment as seen by the unit under test.

goeland86, 2019-01-27:
If you have one insert point for configuration parameters that provides an entire printer scope of parameters, what harm can there be in providing additional parameters where the scope under test is narrower? I'm not saying we provide useless parameters, but rather that we have each test fetch the parameters it needs from that global scope to be able to run.

Wackerbarth, 2019-01-27:
Because inserting any "entire scope" never isolates the code from portions of the code. As a contrived example, you could not design a test that shows that probing is not affected by the OP interface. This is because the "entire scope" must include that interface.