intelligent-agent/redeem #178: Soft Endstops Should Be Checked Before Bed Leveling Transformation

ThatWileyGuy, 2019-02-06:
As the title says, Redeem should be running soft endstop checks before applying the bed leveling matrix. The soft endstops will always be in world-space, not bed-space.

Wackerbarth, 2019-02-06:
I disagree. The soft end stops should serve as limits in the workpiece space to detect objects which do not conform to the operational design of the printer workspace. Bed leveling may require that the specified space be smaller than some physical limits of the mechanism. But the allowable limit should not vary from run to run.

ThatWileyGuy, 2019-02-06:
... yes?
All I'm saying is that the soft endstop positions are defined in the same coordinate space as the hard endstops, which I'm calling "world space." They need to be evaluated in that space; they do not have meaning once coordinates have been transformed via the bed leveling matrix to get "bed space."

If my X axis ends 200mm away from the X=0 endstop, it doesn't matter whether I command a move to X=201 with no bed leveling matrix or if I command a move to X=190 but the bed matrix transforms it to X=201. The point is unreachable.

Wackerbarth, 2019-02-06:
We view those soft end stops differently. You want it to act as a virtual limit switch to replace a missing physical switch and limit the actuator motion. On the other hand, I look for it to define a "working volume" wherein all points are reachable. With a tilted bed, the distinction is important, for example, in the Z-axis where the mechanism has different limits based on the horizontal position and the limit is not the limit of the actuator but rather the position of an obstacle.

ThatWileyGuy, 2019-03-31:
You're right and I believe this code is currently correct.

closed 2019-03-31 by ThatWileyGuy