intelligent-agent/redeem #181: 2.2.1-RC A4A boards not setting current properly

zaped212, 2019-02-25:
I did a clean re-image of 2.2.1-RC image, uploaded my config file through the octoprint ui, and manually created the link between my config and the printer.cfg file ( current octoprint integration looks broken ).

I have my motor current set to 0.8, xyz microstepping at 3

Though I am still seeing the same issue where F values < 300 or so work, but anything faster doesnt.

I do notice that the motor hold strength is very very very weak.

when I tried to move things on the old image I would end up moving the whole printer. now I can move everything without much issue.

Motors were definitely "on" as I hit "motors off" and they became even easier to move.

I used M907 to change the current values from 0, 0.3, 1.0, and 1.5 ( verified via the logs. ex: Setting voltage to 0.0, and Setting voltage to 0.75 ), and tried to move the sled on the tower.

Unfortunately I did not feel any change in force at the different values.
Looking in Stepper.py I notice that B1,B2,B3 all use PWM_DAC while the A4A is using hardware DAC wonder if 2.2.1 doesnt support the hardware DAC

zaped212, 2019-02-27:
Just a quick update.
Opened up my printer and took a look at the AVREF, and BVREF pins of the stepper driver.
Was reading 2.5mV
Changed the current to 1.5A

From the logs:
`
DEBUG    Executing M907 from unbuffered M907 X1.5 Y1.5 Z1.5
DEBUG    Setting voltage to 0.75
DEBUG    Setting voltage to 0.75
DEBUG    Setting voltage to 0.75
DEBUG    Completed M907 from unbuffered M907 X1.5 Y1.5 Z1.5
`
Re-measured the pins and still measured 2.5mV so looks like the DAC isnt being set properly.
Will continue to look into it.

goeland86, 2019-02-27:
Follow up details requested from @zaped212 :
Kamikaze 2.0.8, we have `Adafruit-BBIO (0.0.30)`
Umikaze 2.1.1 we have `Adafruit-BBIO (1.0.4)`

zaped212, 2019-02-28:
Tested 0.0.30, and 1.0.4 an neither of them resolved the issue.
Infact both of them had problems running. Had to comment out the GPIO lines in stepper.py to even be able to set the current and that didnt even work.
I diff'd 1.0.4 to 1.1.1 and didnt see too much in the GPIO and SPI parts of the code that have changed / would cause problems.

zaped212, 2019-03-01:
So I think the issue is that we are no longer "binding" to the correct spidev device.
There was an Adafruit BBIO change with how the channel / device is selected.

Previously it looked like SPI(0,0) would actaully try and connect to spidev2_0
With the new version of code it looks like it probably is connecting to spidev0_0

When I look at "ls /dev/spi*" it shows:
`
root@kamikaze:~# ls /dev/spi*
/dev/spidev2.0  /dev/spidev2.1
`
But with Kamikaze I am seeing 
`
/dev/spidev1.0  /dev/spidev1.1 /dev/spidev2.0  /dev/spidev2.1
`

Ultimately I am not sure what the 1.0 or 1.1 are from in the 2.2.1-RC2 build

goeland86, 2019-03-01:
That may actually be related to a change in numbering the devices based on
the kernel image running.
Or a change in udev rules.

zaped212, 2019-03-01:
https://github.com/intelligent-agent/redeem/pull/182

goeland86, 2019-03-10:
@zaped212 since your PR was merged, I believe this issue can be safely closed?

zaped212, 2019-03-10:
I can agree to that

zaped212, 2019-03-11:
PR #182 was merged to address this.

closed 2019-03-11 by zaped212