intelligent-agent/redeem #183: Printer locks up during home command

zaped212, 2019-03-03:
I have noticed several times that my printer ( custom built deltabot ) can lockup in the middle of a G28 homing command.

From the logs:
2019-03-03 19:48:47,291 root         DEBUG    Coarse search done!
2019-03-03 19:48:47,910 root         INFO     End Stop Z1 hit!
2019-03-03 19:48:48,096 root         INFO     End Stop X1 hit!
2019-03-03 19:48:48,106 root         INFO     End Stop Y1 hit!
2019-03-03 19:49:09,422 root         DEBUG    Poking watchdog
2019-03-03 19:49:39,486 root         DEBUG    Poking watchdog
2019-03-03 19:50:09,547 root         DEBUG    Poking watchdog
2019-03-03 19:50:39,608 root         DEBUG    Poking watchdog
2019-03-03 19:51:09,673 root         DEBUG    Poking watchdog
2019-03-03 19:51:39,738 root         DEBUG    Poking watchdog
2019-03-03 19:52:09,814 root         DEBUG    Poking watchdog
2019-03-03 19:52:39,869 root         DEBUG    Poking watchdog
2019-03-03 19:53:09,929 root         DEBUG    Poking watchdog
2019-03-03 19:53:39,991 root         DEBUG    Poking watchdog

From the logs it looks like the coarse search finishes successfully, but the machine never performs the backoff step before performing the fine search.

I previously added additional logs ( no longer have them ) that showed the system waiting for the fine search to succeed.

The interesting part though, that log showed that the backoff finished successfully, but the printer never moved to back off

ThatWileyGuy, 2019-03-04:
I have observed this and forced a process dump to try and diagnose it, but I didn't come away with a good explanation yet.

The logs are a bit misleading here - the backoff move is getting queued successfully but is never executing. The path planner is waiting for an interrupt from the PRU telling it the coarse seek is complete, but that interrupt seemingly never comes.

ThatWileyGuy, 2019-03-04:
Do you have a consistent way to make this occur or is it just random?
Is it always on the first homing attempt after restarting Redeem?

zaped212, 2019-03-05:
For some reason on my delta I would tend to see it more often after running G33 N4 -> N8.

Wasnt sure if 2.2.1 would help prevent the problem, or if it would help me diagnose it.

It always seemed like a race condition to me, where signals were getting sent but interrupted.

@ThatWileyGuy  can you provide me with instructions on how to perform the process dump?
Ive done a bit of multithreading work at work and would be happy to help dig into this.

Been a few too many times where the printer has locked itself up when I tell it to print, and come back a bit later and its still homing with everything heated.

Gearshredder, 2019-06-30:
It's random for me.