intelligent-agent/redeem #187: Serial Error on start of print

raabomat, 2019-03-24:
[SerialError.zip](https://github.com/intelligent-agent/redeem/files/3000878/SerialError.zip)

With the attached gcode and redeem config i get a serial error. Terminal log:

`Changing monitoring state from "Operational" to "Printing from SD"
Send: M23 /lcl/ae10_3dbenchy.gcodeRecv: File opened:/usr/share/models/AE10_3DBenchy.gcode Size:4786997
Recv: File selected[...]
Send: M24[...]
Recv: ok : M24 in progress[...]
Recv: End stop Y1 hit![...]
Recv: End stop X1 hit!
Recv: End stop Y1 hit!
Recv: End stop X1 hit![...]
Recv: End stop Z1 hit![...]
Recv: End stop Z1 hit![...]
Unexpected error while writing to serial port: SerialException: 'write failed: [Errno 5] Input/output error' @ comm.py:_do_send_without_checksum:3228Please see https://faq.octoprint.org/serialerror for possible reasons of this.Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'read failed: [Errno 5] Input/output error' @ comm.py:_readline:2605Please see https://faq.octoprint.org/serialerror for possible reasons of this.Connection closed, closing down monitorChanging monitoring state from "Printing from SD" to "Offline (Error: SerialException: 'read failed: [Errno 5] Input/output error' @ comm.py:_readline:2605)"Closing down send loop`

ThatWileyGuy, 2019-03-26:
Looks like a crash in the path planner. I'm debugging it.

You might want to look into your start script, though - `G1 Z15.0 F{travel_speed} ;move the platform down 15mm` doesn't seem to be what you wanted :)

ThatWileyGuy, 2019-03-26:
Actually, looks like that's the entire issue - that line is managing to set the move speed to 0 and then try to move the Z axis, and the path planner gets upset because the move will take infinitely long.

There are actually two of these - you'll want to fix both.
```
G1 Z15.0 F{travel_speed} ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F{travel_speed}
```

ThatWileyGuy, 2019-03-26:
Retitling the issue because it's a bug we'll want to address...