intelligent-agent/redeem #188: M909 doesn't actually update the steps per mm and doesn't support microstepping mode 8

ThatWileyGuy, 2019-03-28:
(see the title)

Wackerbarth, 2019-03-28:
I'm not surprised. This is a configuration item that I suspect most users set in their configuration file and never use the M-code.  There are likely to be significant difficulties in changing the value without resetting the printer. Is there a need and is it really worth the effort to support this as an M-code?

goeland86, 2019-03-28:
The desire to support this as an M-code stems from the desire to be able to have a configuration wizard that sets up Redeem as you open up OctoPrint on a fresh install for the first time - perhaps a limited scope, but that would be a desirable use-case.

Another use-case would potentially be fine-tuning extruder steps-per-mm for slightly different diameter filaments that you don't want to adjust in the slicer profile, but I would tend to recommend tweaking slicer settings instead.

Wackerbarth, 2019-03-28:
OctoPrint already provides a mechanism to alter the configuration files. That pathway could be used by some "wizard" instead of trying to force it through the motion-code interface.

As for the extruders, the fine tuning is usually handled by `M221`. Thus, any need to changes to the base steps-per-mm would revert to the same usage case as that of initial setup.

It should also be noted that,  in Marlin, `M909` is a totally unrelated command.

commit 4b0a83fc (2019-08-24): Fix M909 (set microstepping) and M92 (set steps per mm)

- remove Printer.steps_pr_meter because it's redundant with Stepper.microsteps
- remove callback from Stepper to Printer to update steps_pr_meter
- remove redundant microstepping bounds check from M909 - the Steppers can do this themselves
- add microstepping bounds check to B-series Steppers to match the A-series
- update steps per mm in M92 correctly (instead of restarting the path planner, which doesn't help)
- add tests for M909 and M92
- revise PathPlanner::setAxisStepsPerMeter to preserve the current location of the printer

Valida [... cut]
closed 2019-08-24 by ThatWileyGuy