intelligent-agent/redeem #190: Redeem does not crash immediately when it doesn't detect a Cape

goeland86, 2019-03-31:
This particular `else` should make Redeem fail immediately, to clarify why it failed to the user.

https://github.com/intelligent-agent/redeem/blob/00071ca487fae4ca07027e9e84662b8a31e7dece/redeem/Redeem.py#L147

Wackerbarth, 2019-04-01:
I don't think that your suggestion is the appropriate behavior. Redeem should never "crash".  Since it is chosen to use OP as the user interface, when Redeem is not properly configured either because of failure to detect appropriate hardware, or a software misconfiguration, it should still interact with OP and deliver error responses to any commands received.

goeland86, 2019-04-01:
Perhaps crashing isn't the right action, but it shouldn't pretend it has a replicape attached to it - as @ThatWileyGuy describes it, it will crash later on. Changing this to crash immediately at least means that a user will see that the cape not being detected is the main reason it failed to start, instead of later when it tries to address an I2C device that isn't present.